josevcm/nfc-laboratory #3: Install Issue with nfc-lab-2.1.0-x86_64.exe on Windows10Pro 64bits 21H1

bvernoux, 2021-09-25:
I have just tried the latest official release of nfc-lab v2.1.0 from https://github.com/josevcm/nfc-laboratory/releases/download/2.1.0/nfc-lab-2.1.0-x86_64.exe and it does not start correctly.
The installation is fine on Windows10Pro 64bits 21H1 but when I execute nfc-lab.exe it start and exit immediately (it was same issue with previous version https://github.com/josevcm/nfc-laboratory/releases/download/2.0.0/nfc-lab-2.0.0-x86_64.exe)
For information I have installed it in default path `C:\Program Files\nfc-laboratory`
When I try to debug it with Visual Studio Community 2019 I obtain an Illegal Instruction on mufft_create_plan_1d_c2c(void) (see nfc-laboratory/src/nfc-lib/lib-ext/mufft/src/main/cpp\fft.c Line 628 => mufft_create_plan_1d_c2c(void) in Call Stack)
--- D:/workspace/github/nfc-laboratory/src/nfc-lib/lib-ext/mufft/src/main/cpp\fft.c 
00000000004C1D40 C4 E2 78 F3 C9       blsr        eax,ecx  
For more details see:
[image]

For information/reference when I'm doing my own build using this updated script https://github.com/bvernoux/nfc-laboratory#build-from-qtcreator (see it in https://github.com/josevcm/nfc-laboratory/pull/2) all work fine
See my binaries build if you want to check it on your computer:
- https://hydrabus.com/nfc-labv2_1_0_ReleaseBuild_QtCreator5_0_1-Qt_5_15_2-mingw81_64.7z
- `nfc-labv2_1_0_ReleaseBuild_QtCreator5_0_1-Qt_5_15_2-mingw81_64.7z`
   - SHA256: FDC191F73D2B06A0C1F35CFF7AC64F58C8707DFF03E36C38575751E26E48220B

josevcm, 2021-09-25:
Thanks, I will fix installer as son as posible

bvernoux, 2021-10-21:
We can close this issue as latest release https://github.com/josevcm/nfc-laboratory/releases/tag/2.4.0 have fixed that issue