josevcm/nfc-laboratory #4: nfc-lab v2.1.0 "Filter/Setup"seems to be not implemented

bvernoux, 2021-09-25:
With nfc-lab v2.1.0 (my own build see https://github.com/josevcm/nfc-laboratory/issues/3) It seems when clicking on the Filter there is no effect (and I do not see where to enter the filter and what are the rules ?)
[image]

Also Setup have no effect when we click on it
[image]

josevcm, 2021-09-25:
Hi Benjamin

I have to add the user manual....

The filtering function serves to limit the number of identical messages
that are received, it only acts if  is received repeated message many times.

The settings is not implemented yet, sorry!

I will fix at son as posible

bvernoux, 2021-09-25:
About the Filter:
- Maybe it it will be great to add a real filter supporting also regExp PCRE (like the one supported by default in Qt => QRegExp) ?
- I have done that in my similar NFC sniffer tool see https://github.com/hydrabus/hydratool

josevcm, 2021-09-25:
Hi Benjamin,

Regexp filtering it's a good improvement, I will add to roadmap.

josevcm, 2022-05-29:
Hi

Already included in development branch, release soon...

[image]

closed 2022-05-29 by josevcm