josevcm/nfc-laboratory #10: nfc-lab v2.6.6 build broken with QtCreator5.0.2 / MinGW 8.1

bvernoux, 2022-01-16:
Since nfc-lab v2.6.6 latest commit the build is broken with QtCreator5.0.2 / MinGW 8.1 (because of change to support MinGW 9.0 to be clarified)
It will be great to support if possible MinGW 8.1(by default with latest version of QtCreator5.0.2) and the new MinGW 9.0

josevcm, 2022-01-16:
Hi Benjamin

MinGW 8.1 has a bug in the <filesystem> file that prevents its use, see https://sourceforge.net/p/mingw-w64/bugs/737/.

I have replaced it with my own implementation to be able to compile with 8.1 versions, you can find the fix in the last commit.

Regards

bvernoux, 2022-01-16:
Thanks all work fine now with your fix https://github.com/josevcm/nfc-laboratory/commit/64645adacdafc9e1f280b22c2fb927d712c190b7 and when using QtCreator5.0.2 / MinGW 8.1

closed 2022-01-16 by bvernoux