josevcm/nfc-laboratory #14: I-Block / S-Block mismatch

gentilkiwi, 2022-06-11:
Hello,

On version 2.8.0, it seems a `S-Block` is interpreted as `I-Block` when receiving from an `I-Block`
On the oppositive way, when receiving an `I-Block` after a `S-Block`, it's not interpreted.

No problem on main listing, but protocol decoding seems to be incorrect.

[image]
[image]

josevcm, 2022-06-12:
Hi Benjamin

Thanks for the information, I just corrected it on the development branch, I will generate a release soon.

Regards

closed 2022-06-12 by josevcm

gentilkiwi, 2022-06-12:
Thank you Jose!

[image]