josevcm/nfc-laboratory #21: may i get the wave data?

yywei0323, 2023-05-09:
May I ask how can i get the wave data? I found the wave data of "ATQA" ，I wonder if i can get my devices wave data and how.

josevcm, 2023-05-09:
Hi

By pressing recording buttom you can get raw wave data, the file will be
created insiste installation directory.

 Regards

yywei0323, 2023-05-10:
I 'm sorry to bother you again. I have tried to press the recording bottom on the left of the tool box. There isn't the wave file in the installation directory as the picture. Besides, after pressing the bottom, i can't presss any bottom of the tool box. Did i operate incorrectly?
[image]

josevcm, 2023-05-10:
The "Live" and "Record" buttons are disabled, does the Airspy device work correctly? 
If you have installed the application in "C:\Program Files" it is possible that windows blocks access to disk and does not allow recording, try installing it in another folder.

yywei0323, 2023-05-10:
Thanks for your in time response ! ！！ (Bow!) 
I have got the wave file. I have filted the ATQA data, and recorded it. Does it mean the signal data in the file only match the "ATQA" command? If not, can i use some method to get the data?
[image]

josevcm, 2023-05-10:
You can zoom-out all capture using mouse well to see complete signal. Your screenshot show REQA in the left, but for some reason is not correctly decoded, you should see the that command at the top, you can try different distances from the antenna and card to get a clear signal.

josevcm, 2023-05-10:
> I understand. I need to analyze the signal only contain the "ATQA"，which
> is blue in the signal zoom without the other signal, can i get the filted
> wave data? Can I get the data from nfc-laboratory, or could you please tell
> how to get the data with other tools?

NFC-LAB does not allow you to cut pieces of the signal, for that you can
open the complete file with Audaticy (https://www.audacityteam.org/) and
extract the parts you need.

closed 2023-05-20 by josevcm