josevcm/nfc-laboratory #22: Linux support

olafium, 2023-05-30:
Found this project mentioned on [Hackaday](https://hackaday.com/2023/05/27/hacking-a-smart-electric-toothbrush-to-reset-its-usage-counter/), congratulations. I wish to try it out myself, but sadly I don't use Windows.

I'm aware it's windows-only, but I tried my luck with compiling it on Linux (QT Creator 9.0.2) and the error was following:

`src/nfc-lib/lib-rt/rt-lang/src/main/cpp/Logger.cpp:241: error: ‘localtime_s’ was not declared in this scope; did you mean ‘localtime_r’?
src/nfc-lib/lib-rt/rt-lang/src/main/cpp/Logger.cpp: In member function ‘void rt::LogWriter::write(rt::LogEvent*)’:
src/nfc-lib/lib-rt/rt-lang/src/main/cpp/Logger.cpp:241:7: error: ‘localtime_s’ was not declared in this scope; did you mean ‘localtime_r’?
  241 |       localtime_s(&timeinfo, &seconds);
      |       ^~~~~~~~~~~
      |       localtime_r`

Since I'm not big on C, I have no idea how difficult it would be. However I am willing to test out any possible future developments!

Cheers

josevcm, 2023-05-30:
Hello,

This line is for logging purposes only, try commenting it out with // and recompile.

// localtime_s(&timeinfo, &seconds);

Right now I don't have Linux Machiche to verify this.

josevcm, 2023-06-01:
I'm very sorry for the inconvenience, this project was designed for X86 architecture, I don't have a MAC to test it and see if it's possible to fix it.

josevcm, 2023-06-04:
> Found this project mentioned on [Hackaday](https://hackaday.com/2023/05/27/hacking-a-smart-electric-toothbrush-to-reset-its-usage-counter/), congratulations. I wish to try it out myself, but sadly I don't use Windows.
> 
> I'm aware it's windows-only, but I tried my luck with compiling it on Linux (QT Creator 9.0.2) and the error was following:
> 
> ```
> src/nfc-lib/lib-rt/rt-lang/src/main/cpp/Logger.cpp:241: error: ‘localtime_s’ was not declared in this scope; did you mean ‘localtime_r’?
> src/nfc-lib/lib-rt/rt-lang/src/main/cpp/Logger.cpp: In member function ‘void rt::LogWriter::write(rt::LogEvent*)’:
> src/nfc-lib/lib-rt/rt-lang/src/main/cpp/Logger.cpp:241:7: error: ‘localtime_s’ was not declared in this scope; did you mean ‘localtime_r’?
>   241 |       localtime_s(&timeinfo, &seconds);
>       |       ^~~~~~~~~~~
>       |       localtime_r
> ```
> 
> Since I'm not big on C, I have no idea how difficult it would be. However I am willing to test out any possible future developments. (If it matters, the exact distro is debian sid)
> 
> Cheers

Hi again

Please check the development branch again, I added some changes to support linux build, but it was only tested with my WSL2 distro on windows, I don't know if it would work well on native linux.

josevcm, 2023-06-06:
Thanks for trying it on MAC Tom, for the moment I will disable the SMID instructions so that they only apply on x86. 

The FFT display is not the main functionality of nfc-laboratory, and really not very useful I think. Select muFFT because of its simplicity and easy compilation, plus the MIT license.

josevcm, 2023-06-18:
Sorry for the delay, I have not yet been able to review your comments or the PR to integrate it, I have not forgotten. I'll do it when I have time.

josevcm, 2023-06-22:
> ```
> target_compile_options(airspy PRIVATE "-msse2" -DUSE_SSE2)
> ```
> 
> in `src/nfc-lib/lib-ext/airspy/CMakeLists.txt` also needs to be disabled for Arm

Please Tom, check again, I fixed the related problems with Linux Build in develop branch

Thanks

closed 2023-06-30 by josevcm