josevcm/nfc-laboratory #29: TransacB - one frame not decoded

chrisrock1984, 2023-10-25:
when running transacB polling, the frame before the last is not decoded but clearly visible in the signal windows.
decoding itself runs perfect, except after ATTRIB the decoder seems to be "blind" for some time.
[image]
[image]

josevcm, 2023-10-25:
Hi

Yes, I know this case, the message is not detected because I did not find the specification for that modulation, it may be a variation of type B, but it did not meet the characteristics of the documents that appears in /doc/pdf.

If you know what it is, it would help me identify it correctly.

Thanks!

closed 2023-11-01 by josevcm

josevcm, 2024-11-19:
This issue is already fixed in New versión, please check again