josevcm/nfc-laboratory #31: Dir-NOTFOUND LIBUSB

dose13seitz, 2024-01-04:
Hi,

I am running Ubuntu 20.04 and it seems that some references were not found. This is the error message that I got. 

Thank's for your help!

dose@thinkpad:~$ sudo cmake -DCMAKE_BUILD_TYPE=Release -S nfc-laboratory -B cmake-build-release
-- Build for x86_64
-- Enabled SSE/SSE3 instruction set
-- Detected libusb LIBUSB_INCLUDE_DIR-NOTFOUND LIBUSB_LIBRARY-NOTFOUND
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/home/dose/nfc-laboratory/src/nfc-app/app-qt/LIBUSB_INCLUDE_DIR
   used as include directory in directory /home/dose/nfc-laboratory/src/nfc-app/app-qt
/home/dose/nfc-laboratory/src/nfc-lib/lib-ext/airspy-tools/LIBUSB_INCLUDE_DIR
   used as include directory in directory /home/dose/nfc-laboratory/src/nfc-lib/lib-ext/airspy-tools
/home/dose/nfc-laboratory/src/nfc-lib/lib-ext/airspy/LIBUSB_INCLUDE_DIR
   used as include directory in directory /home/dose/nfc-laboratory/src/nfc-lib/lib-ext/airspy
/home/dose/nfc-laboratory/src/nfc-lib/lib-ext/crapto1/LIBUSB_INCLUDE_DIR
   used as include directory in directory /home/dose/nfc-laboratory/src/nfc-lib/lib-ext/crapto1
/home/dose/nfc-laboratory/src/nfc-lib/lib-ext/rtlsdr/LIBUSB_INCLUDE_DIR
   used as include directory in directory /home/dose/nfc-laboratory/src/nfc-lib/lib-ext/rtlsdr
LIBUSB_LIBRARY
    linked by target "airspy" in directory /home/dose/nfc-laboratory/src/nfc-lib/lib-ext/airspy
    linked by target "airspy-rx" in directory /home/dose/nfc-laboratory/src/nfc-lib/lib-ext/airspy-tools
    linked by target "rtlsdr" in directory /home/dose/nfc-laboratory/src/nfc-lib/lib-ext/rtlsdr

-- Configuring incomplete, errors occurred!
See also "/home/dose/cmake-build-release/CMakeFiles/CMakeOutput.log".

dose13seitz, 2024-01-05:
Hi,

Thank's for your reply. I installed the above package but it is still not working. With regard to libusb, I have the following packages installed now:

dose@thinkpad:~$ apt list --installed | grep libusb*

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libhidapi-libusb0/focal,now 0.9.0+dfsg-1 amd64  [Installiert,automatisch]
libusb-0.1-4/focal,now 2:0.1.12-32 amd64  [Installiert,automatisch]
libusb-1.0-0/focal,now 2:1.0.23-2build1 amd64  [installiert]
libusb-dev/focal,now 2:0.1.12-32 amd64  [installiert]
libusbmuxd6/focal,now 2.0.1-2 amd64  [Installiert,automatisch]

dpkg -L shows the follwoing:

dose@thinkpad:~$ dpkg -L libusb-1.0-0 
/.
/lib
/lib/x86_64-linux-gnu
/lib/x86_64-linux-gnu/libusb-1.0.so.0.2.0
/usr
/usr/share
/usr/share/doc
/usr/share/doc/libusb-1.0-0
/usr/share/doc/libusb-1.0-0/README
/usr/share/doc/libusb-1.0-0/changelog.Debian.gz
/usr/share/doc/libusb-1.0-0/copyright
/lib/x86_64-linux-gnu/libusb-1.0.so.0

and

dose@thinkpad:~$ dpkg -L libusb-dev 
/.
/usr
/usr/bin
/usr/bin/libusb-config
/usr/include
/usr/include/usb.h
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libusb.a
/usr/lib/x86_64-linux-gnu/libusb.so
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/libusb.pc
/usr/share
/usr/share/doc
/usr/share/doc/libusb-dev
/usr/share/doc/libusb-dev/copyright
/usr/share/doc/libusb-dev/html
/usr/share/doc/libusb-dev/html/api-device-interfaces.html
/usr/share/doc/libusb-dev/html/api-return-values.html
/usr/share/doc/libusb-dev/html/api-synchronous.html
/usr/share/doc/libusb-dev/html/api-timeouts.html
/usr/share/doc/libusb-dev/html/api-types.html
/usr/share/doc/libusb-dev/html/api.html
/usr/share/doc/libusb-dev/html/examples-code.html
/usr/share/doc/libusb-dev/html/examples-other.html
/usr/share/doc/libusb-dev/html/examples-tests.html
/usr/share/doc/libusb-dev/html/examples.html
/usr/share/doc/libusb-dev/html/function.usbbulkread.html
/usr/share/doc/libusb-dev/html/function.usbbulkwrite.html
/usr/share/doc/libusb-dev/html/function.usbclaiminterface.html
/usr/share/doc/libusb-dev/html/function.usbclearhalt.html
/usr/share/doc/libusb-dev/html/function.usbclose.html
/usr/share/doc/libusb-dev/html/function.usbcontrolmsg.html
/usr/share/doc/libusb-dev/html/function.usbdetachkerneldrivernp.html
/usr/share/doc/libusb-dev/html/function.usbfindbusses.html
/usr/share/doc/libusb-dev/html/function.usbfinddevices.html
/usr/share/doc/libusb-dev/html/function.usbgetbusses.html
/usr/share/doc/libusb-dev/html/function.usbgetdescriptor.html
/usr/share/doc/libusb-dev/html/function.usbgetdescriptorbyendpoint.html
/usr/share/doc/libusb-dev/html/function.usbgetdrivernp.html
/usr/share/doc/libusb-dev/html/function.usbgetstring.html
/usr/share/doc/libusb-dev/html/function.usbgetstringsimple.html
/usr/share/doc/libusb-dev/html/function.usbinit.html
/usr/share/doc/libusb-dev/html/function.usbinterruptread.html
/usr/share/doc/libusb-dev/html/function.usbinterruptwrite.html
/usr/share/doc/libusb-dev/html/function.usbopen.html
/usr/share/doc/libusb-dev/html/function.usbreleaseinterface.html
/usr/share/doc/libusb-dev/html/function.usbreset.html
/usr/share/doc/libusb-dev/html/function.usbresetep.html
/usr/share/doc/libusb-dev/html/function.usbsetaltinterface.html
/usr/share/doc/libusb-dev/html/function.usbsetconfiguration.html
/usr/share/doc/libusb-dev/html/functions.html
/usr/share/doc/libusb-dev/html/index.html
/usr/share/doc/libusb-dev/html/intro-overview.html
/usr/share/doc/libusb-dev/html/intro-support.html
/usr/share/doc/libusb-dev/html/intro.html
/usr/share/doc/libusb-dev/html/preface.html
/usr/share/doc/libusb-dev/html/ref.bulk.html
/usr/share/doc/libusb-dev/html/ref.control.html
/usr/share/doc/libusb-dev/html/ref.core.html
/usr/share/doc/libusb-dev/html/ref.deviceops.html
/usr/share/doc/libusb-dev/html/ref.interrupt.html
/usr/share/doc/libusb-dev/html/ref.nonportable.html
/usr/share/doc-base
/usr/share/doc-base/libusb-dev
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/libusb-config.1.gz
/usr/share/doc/libusb-dev/changelog.Debian.gz

Any ideas?

josevcm, 2024-01-05:
Please check if you have installed libusb*-1.0-0-**dev *and try again

dose13seitz, 2024-01-05:
That did the trick! Many Thank's for your help!

closed 2024-01-05 by dose13seitz