josevcm/nfc-laboratory #39: Problem vith build on Manjaro Linux

iphizic, 2024-11-17:
Hello, I try build nfc-laboratory on manjaro and have this error:
```
[ 25%] Built target mufft
[ 26%] Building CXX object src/nfc-lib/lib-rt/rt-lang/CMakeFiles/rt-lang.dir/src/main/cpp/Tokenizer.cpp.o
[ 26%] Building CXX object src/nfc-lib/lib-rt/rt-lang/CMakeFiles/rt-lang.dir/src/main/cpp/Logger.cpp.o
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/cpp/Logger.cpp: In function ‘int rt::getLevelIndex(std::string)’:
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/cpp/Logger.cpp:58:9: error: ‘transform’ is not a member of ‘std’
   58 |    std::transform(name.begin(), name.end(), name.begin(), toupper);
      |         ^~~~~~~~~
make[3]: *** [src/nfc-lib/lib-rt/rt-lang/CMakeFiles/rt-lang.dir/build.make:174: src/nfc-lib/lib-rt/rt-lang/CMakeFiles/rt-lang.dir/src/main/cpp/Logger.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:1258: src/nfc-lib/lib-rt/rt-lang/CMakeFiles/rt-lang.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:581: src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/rule] Error 2
make: *** [Makefile:124: nfc-spy] Error 2
```

josevcm, 2024-11-17:
what version of gcc are you using? must be 11 or higer, check this output when configure CMAKE:

user@LG:~$ cmake -DCMAKE_BUILD_TYPE=Release -S nfc-laboratory -B cmake-build-release
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0

iphizic, 2024-11-18:
Configure CMAKE output:
```
$ cmake -DCMAKE_BUILD_TYPE=Release -S nfc-laboratory -B cmake-build-release
-- The C compiler identification is GNU 14.2.1
-- The CXX compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build for x86_64
-- Enabled SSE/SSE3 instruction set
-- usb-1.0:
--   include:/usr/include/libusb-1.0
--   library:/usr/lib/libusb-1.0.so
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found WrapAtomic: TRUE
-- Found OpenGL: /usr/lib/libOpenGL.so
-- Found WrapOpenGL: TRUE
-- Found XKB: /usr/lib/libxkbcommon.so (found suitable version "1.7.0", minimum required is "0.5.0")
-- Found WrapVulkanHeaders: /usr/include
-- Configuring done (1.8s)
-- Generating done (0.3s)
-- Build files have been written to: /home/iphizic/build/cmake-build-release
```

And now I hawe this errors:
```
[ 27%] Built target rt-lang
[ 27%] Building CXX object src/nfc-lib/lib-hw/hw-dev/CMakeFiles/hw-dev.dir/src/main/cpp/hw/DeviceFactory.cpp.o
[ 27%] Building CXX object src/nfc-lib/lib-hw/hw-dev/CMakeFiles/hw-dev.dir/src/main/cpp/hw/SignalBuffer.cpp.o
[ 29%] Building CXX object src/nfc-lib/lib-hw/hw-dev/CMakeFiles/hw-dev.dir/src/main/cpp/hw/RecordDevice.cpp.o
[ 29%] Building CXX object src/nfc-lib/lib-hw/hw-dev/CMakeFiles/hw-dev.dir/src/main/cpp/usb/Usb.cpp.o
In file included from /home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-dev/src/main/include/hw/SignalBuffer.h:25,
                 from /home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-dev/src/main/cpp/hw/SignalBuffer.cpp:22:
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h: In constructor ‘rt::Buffer<T>::Alloc::Alloc(unsigned int, unsigned int, unsigned int, unsigned int, void*)’:
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h:50:21: error: there are no arguments to ‘malloc’ that depend on a template parameter, so a declaration of ‘malloc’ must be available [-fpermissive]
   50 |             block = malloc(capacity * sizeof(T) + BUFFER_ALIGNMENT);
      |                     ^~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h:50:21: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h: In destructor ‘rt::Buffer<T>::Alloc::~Alloc()’:
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h:59:13: error: there are no arguments to ‘free’ that depend on a template parameter, so a declaration of ‘free’ must be available [-fpermissive]
   59 |             free(block);
      |             ^~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h: In instantiation of ‘rt::Buffer<T>::Alloc::~Alloc() [with T = float]’:
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h:111:13:   required from ‘rt::Buffer<T>::~Buffer() [with T = float]’
  111 |             delete alloc;
      |             ^~~~~~~~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-dev/src/main/cpp/hw/SignalBuffer.cpp:38:71:   required from here
   38 | SignalBuffer::SignalBuffer() : impl(std::make_shared<Impl>(0, 0, 0, 0))
      |                                                                       ^
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h:59:17: error: ‘free’ was not declared in this scope
   59 |             free(block);
      |             ~~~~^~~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h:29:1: note: ‘free’ is defined in header ‘<cstdlib>’; this is probably fixable by adding ‘#include <cstdlib>’
   28 | #include <functional>
  +++ |+#include <cstdlib>
   29 | 
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h: In instantiation of ‘rt::Buffer<T>::Alloc::Alloc(unsigned int, unsigned int, unsigned int, unsigned int, void*) [with T = float]’:
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h:104:186:   required from ‘rt::Buffer<T>::Buffer(unsigned int, unsigned int, unsigned int, unsigned int, void*) [with T = float]’
  104 |       explicit Buffer(unsigned int capacity, unsigned int type = 0, unsigned int stride = 1, unsigned int interleave = 1, void *context = nullptr) : state(0, capacity, capacity), alloc(new Alloc(type, capacity, stride, interleave, context))
      |                                                                                                                                                                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-dev/src/main/cpp/hw/SignalBuffer.cpp:42:339:   required from here
[... cut ...]
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Buffer.h:50:27: note: ‘malloc’ is defined in header ‘<cstdlib>’; this is probably fixable by adding ‘#include <cstdlib>’
make[3]: *** [src/nfc-lib/lib-hw/hw-dev/CMakeFiles/hw-dev.dir/build.make:104: src/nfc-lib/lib-hw/hw-dev/CMakeFiles/hw-dev.dir/src/main/cpp/hw/SignalBuffer.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:1016: src/nfc-lib/lib-hw/hw-dev/CMakeFiles/hw-dev.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:581: src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/rule] Error 2
make: *** [Makefile:124: nfc-spy] Error 2
```

josevcm, 2024-11-18:
It seems to need some includes, your compiler is more sensitive! I will fix it as soon as possible, Thanks for your support,

josevcm, 2024-11-18:
Fixed #include <cstdlib> in Buffer.h

closed 2024-11-18 by josevcm