josevcm/nfc-laboratory #42: Problem with build Majaro linux

iphizic, 2024-11-18:
Hello, I have problem again:
```
[ 32%] Built target hw-radio
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-logic/src/main/cpp/DSLogicDevice.cpp: In member function ‘bool hw::DSLogicDevice::Impl::set(int, const rt::Variant&, int)’:
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-logic/src/main/cpp/DSLogicDevice.cpp:558:22: error: ‘find_if’ is not a member of ‘std’; did you mean ‘find’?
  558 |       auto ch = std::find_if(channels.begin(), channels.end(), [channel](const dsl_channel &c) { return c.index == channel; });
      |                      ^~~~~~~
      |                      find
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-logic/src/main/cpp/DSLogicDevice.cpp:665:25: error: cannot convert ‘<brace-enclosed initializer list>’ to ‘std::vector<std::variant<bool, char, short int, int, long int, long long int, unsigned char, short unsigned int, unsigned int, long unsigned int, long long unsigned int, float, double, char*, void*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::thread::id, std::chrono::duration<long long int, std::ratio<1, 1000000000> >, std::map<int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<int>, std::allocator<std::pair<const int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, rt::Buffer<unsigned char>, std::vector<int, std::allocator<int> > > >’
  665 |                log->info("setting VDIV of channel {} to {} mv", {ch->index, ch->vdiv});
      |                ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-logic/src/main/cpp/DSLogicDevice.cpp:28:
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Logger.h:56:65: note:   initializing argument 2 of ‘void rt::Logger::info(const std::string&, std::vector<std::variant<bool, char, short int, int, long int, long long int, unsigned char, short unsigned int, unsigned int, long unsigned int, long long unsigned int, float, double, char*, void*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::thread::id, std::chrono::duration<long long int, std::ratio<1, 1000000000> >, std::map<int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<int>, std::allocator<std::pair<const int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, rt::Buffer<unsigned char>, std::vector<int, std::allocator<int> > > >) const’
   56 |       void info(const std::string &format, std::vector<Variant> params = {}) const;
      |                                            ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-logic/src/main/cpp/DSLogicDevice.cpp:677:25: error: cannot convert ‘<brace-enclosed initializer list>’ to ‘std::vector<std::variant<bool, char, short int, int, long int, long long int, unsigned char, short unsigned int, unsigned int, long unsigned int, long long unsigned int, float, double, char*, void*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::thread::id, std::chrono::duration<long long int, std::ratio<1, 1000000000> >, std::map<int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<int>, std::allocator<std::pair<const int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, rt::Buffer<unsigned char>, std::vector<int, std::allocator<int> > > >’
  677 |                log->info("setting VFACTOR of channel {} to {}", {ch->index, ch->vfactor});
      |                ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Logger.h:56:65: note:   initializing argument 2 of ‘void rt::Logger::info(const std::string&, std::vector<std::variant<bool, char, short int, int, long int, long long int, unsigned char, short unsigned int, unsigned int, long unsigned int, long long unsigned int, float, double, char*, void*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::thread::id, std::chrono::duration<long long int, std::ratio<1, 1000000000> >, std::map<int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<int>, std::allocator<std::pair<const int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, rt::Buffer<unsigned char>, std::vector<int, std::allocator<int> > > >) const’
   56 |       void info(const std::string &format, std::vector<Variant> params = {}) const;
      |                                            ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-logic/src/main/cpp/DSLogicDevice.cpp:693:25: error: cannot convert ‘<brace-enclosed initializer list>’ to ‘std::vector<std::variant<bool, char, short int, int, long int, long long int, unsigned char, short unsigned int, unsigned int, long unsigned int, long long unsigned int, float, double, char*, void*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::thread::id, std::chrono::duration<long long int, std::ratio<1, 1000000000> >, std::map<int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<int>, std::allocator<std::pair<const int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, rt::Buffer<unsigned char>, std::vector<int, std::allocator<int> > > >’
  693 |                log->info("setting coupling of channel {} to {}", {ch->index, ch->coupling});
      |                ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Logger.h:56:65: note:   initializing argument 2 of ‘void rt::Logger::info(const std::string&, std::vector<std::variant<bool, char, short int, int, long int, long long int, unsigned char, short unsigned int, unsigned int, long unsigned int, long long unsigned int, float, double, char*, void*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::thread::id, std::chrono::duration<long long int, std::ratio<1, 1000000000> >, std::map<int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<int>, std::allocator<std::pair<const int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, rt::Buffer<unsigned char>, std::vector<int, std::allocator<int> > > >) const’
   56 |       void info(const std::string &format, std::vector<Variant> params = {}) const;
      |                                            ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-logic/src/main/cpp/DSLogicDevice.cpp:713:37: error: ‘count_if’ is not a member of ‘std’
  713 |                validChannels = std::count_if(channels.begin(), channels.end(), [](const dsl_channel &c) { return c.enabled; });
      |                                     ^~~~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-hw/hw-logic/src/main/cpp/DSLogicDevice.cpp:715:25: error: cannot convert ‘<brace-enclosed initializer list>’ to ‘std::vector<std::variant<bool, char, short int, int, long int, long long int, unsigned char, short unsigned int, unsigned int, long unsigned int, long long unsigned int, float, double, char*, void*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::thread::id, std::chrono::duration<long long int, std::ratio<1, 1000000000> >, std::map<int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<int>, std::allocator<std::pair<const int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, rt::Buffer<unsigned char>, std::vector<int, std::allocator<int> > > >’
  715 |                log->info("setting channel {} enabled to {}", {ch->index, ch->enabled});
      |                ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/iphizic/build/nfc-laboratory/src/nfc-lib/lib-rt/rt-lang/src/main/include/rt/Logger.h:56:65: note:   initializing argument 2 of ‘void rt::Logger::info(const std::string&, std::vector<std::variant<bool, char, short int, int, long int, lo [... cut ...] _string<char, std::char_traits<char>, std::allocator<char> > > > >, rt::Buffer<unsigned char>, std::vector<int, std::allocator<int> > > >) const’
   56 |       void info(const std::string &format, std::vector<Variant> params = {}) const;
      |                                            ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
make[3]: *** [src/nfc-lib/lib-hw/hw-logic/CMakeFiles/hw-logic.dir/build.make:76: src/nfc-lib/lib-hw/hw-logic/CMakeFiles/hw-logic.dir/src/main/cpp/DSLogicDevice.cpp.o] Error 1
```

josevcm, 2024-11-18:
I'm sorry but my gcc 11.4 compiler does not generate those errors, can you try it with a gcc 11 or gcc 13 version? Are you using any compilation flags in environment variables that can modify the behavior? this is my full output to help compare:

```
user@LG:~/github$ cmake -DCMAKE_BUILD_TYPE=Release -S nfc-laboratory -B cmake-build-release
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build for x86_64
-- Enabled SSE/SSE3 instruction set
-- usb-1.0:
--   include:/usr/include/libusb-1.0
--   library:/usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found WrapAtomic: TRUE
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so
-- Found WrapOpenGL: TRUE
-- Could NOT find XKB (missing: XKB_LIBRARY XKB_INCLUDE_DIR) (Required is at least version "0.5.0")
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/github/cmake-build-release

user@LG:~/github$ cmake --build cmake-build-release --target nfc-spy -- -j 10
[  1%] Building C object src/nfc-lib/lib-ext/ed25519/CMakeFiles/ed25519.dir/src/main/c/fe.c.o
[  1%] Building C object src/nfc-lib/lib-ext/ed25519/CMakeFiles/ed25519.dir/src/main/c/add_scalar.c.o
[  1%] Building C object src/nfc-lib/lib-ext/airspy/CMakeFiles/airspy.dir/src/main/c/airspy.c.o
[  2%] Building C object src/nfc-lib/lib-ext/mirisdr/CMakeFiles/mirisdr.dir/src/main/c/libmirisdr.c.o
[  2%] Building C object src/nfc-lib/lib-ext/mufft/CMakeFiles/mufft-avx.dir/src/main/c/x86/kernel.avx.c.o
[  3%] Building C object src/nfc-lib/lib-ext/rtlsdr/CMakeFiles/rtlsdr.dir/src/main/c/librtlsdr.c.o
[  4%] Building C object src/nfc-lib/lib-ext/ed25519/CMakeFiles/ed25519.dir/src/main/c/ge.c.o
[  5%] Building C object src/nfc-lib/lib-ext/mufft/CMakeFiles/mufft-sse3.dir/src/main/c/x86/kernel.sse3.c.o
[  5%] Building C object src/nfc-lib/lib-ext/mufft/CMakeFiles/mufft-sse.dir/src/main/c/x86/kernel.sse.c.o
[  6%] Building C object src/nfc-lib/lib-ext/microtar/CMakeFiles/microtar.dir/src/main/c/microtar.c.o
[  6%] Building C object src/nfc-lib/lib-ext/ed25519/CMakeFiles/ed25519.dir/src/main/c/key_exchange.c.o
[  7%] Building C object src/nfc-lib/lib-ext/ed25519/CMakeFiles/ed25519.dir/src/main/c/keypair.c.o
[  7%] Building C object src/nfc-lib/lib-ext/rtlsdr/CMakeFiles/rtlsdr.dir/src/main/c/tuner_e4k.c.o
[  8%] Linking C static library libmicrotar.a
[  8%] Building C object src/nfc-lib/lib-ext/ed25519/CMakeFiles/ed25519.dir/src/main/c/sc.c.o
[  8%] Built target microtar
[  9%] Building C object src/nfc-lib/lib-ext/ed25519/CMakeFiles/ed25519.dir/src/main/c/seed.c.o
[  9%] Building CXX object src/nfc-lib/lib-rt/rt-lang/CMakeFiles/rt-lang.dir/src/main/cpp/Executor.cpp.o
[ 10%] Building CXX object src/nfc-lib/lib-rt/rt-lang/CMakeFiles/rt-lang.dir/src/main/cpp/FileSystem.cpp.o
[ 11%] Building C object src/nfc-lib/lib-ext/airspy/CMakeFiles/airspy.dir/src/main/c/iqconverter_float.c.o
[ 13%] Building C object src/nfc-lib/lib-ext/rtlsdr/CMakeFiles/rtlsdr.dir/src/main/c/tuner_fc0012.c.o
[ 14%] Linking C static library libmufft-sse.a
[ 14%] Linking C static library libmufft-sse3.a
[ 14%] Building CXX object src/nfc-lib/lib-rt/rt-lang/CMakeFiles/rt-lang.dir/src/main/cpp/Format.cpp.o
[ 14%] Built target mufft-sse
[ 14%] Building C object src/nfc-lib/lib-ext/airspy/CMakeFiles/airspy.dir/src/main/c/iqconverter_int16.c.o
[ 14%] Built target mufft-sse3
[ 15%] Building CXX object src/nfc-lib/lib-rt/rt-lang/CMakeFiles/rt-lang.dir/src/main/cpp/Map.cpp.o
[ 16%] Building C object src/nfc-lib/lib-ext/ed25519/CMakeFiles/ed25519.dir/src/main/c/sha512.c.o
[ 16%] Building C object src/nfc-lib/lib-ext/ed25519/CMakeFiles/ed25519.dir/src/main/c/sign.c.o
[ 17%] Building CXX object src/nfc-lib/lib-rt/rt-lang/CMakeFiles/rt-lang.dir/src/main/cpp/Package.cpp.o
[ 18%] Linking C static library libairspy.a
[... 118 lines cut ...]
[ 85%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/src/main/cpp/widgets/AbstractPlotWidget.cpp.o
[ 85%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/src/main/cpp/widgets/FramesWidget.cpp.o
[ 86%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/src/main/cpp/widgets/HexViewWidget.cpp.o
[ 88%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/src/main/cpp/widgets/FrequencyWidget.cpp.o
[ 88%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/src/main/cpp/widgets/IconDelegate.cpp.o
[ 89%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/src/main/cpp/widgets/ParserDelegate.cpp.o
[ 89%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/src/main/cpp/widgets/ParserWidget.cpp.o
[ 90%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/src/main/cpp/widgets/LogicWidget.cpp.o
[... cut ...]
[ 97%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/nfc-spy_autogen/BZPGUHF7SV/qrc_style.cpp.o
[ 97%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/nfc-spy_autogen/W56BXCATTS/qrc_icons.cpp.o
[ 98%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/nfc-spy_autogen/7WXA5J3H5Z/qrc_style.cpp.o
[ 98%] Building CXX object src/nfc-app/app-qt/CMakeFiles/nfc-spy.dir/nfc-spy_autogen/M2ZWUQEUSF/qrc_app.cpp.o
[100%] Linking CXX executable nfc-spy
[100%] Built target nfc-spy
```

iphizic, 2024-11-18:
I add fixes for close issue

closed 2024-11-18 by iphizic