josevcm/nfc-laboratory #44: RTL-SDR not works

iphizic, 2024-11-19:
I build and run nfc-spy on linux in log see initialization sdr:
```
 Found Rafael Micro R828D tuner
[R82XX] PLL not locked!
Disabled direct sampling mode
[R82XX] PLL not locked!
```
but Acquisition is inactive.

Steffen-W, 2024-11-19:
Same like #41 but for me the code works. Which sdr are you using?

iphizic, 2024-11-19:
Thank you, but, I not see signal. If I run sdrpp signal is present. I use rtl-sdr.com dongle v4.

josevcm, 2024-11-19:
Please set this DEBUG level in the INI file, start the application and send me "nfc-spy.log" to see what happens.

[logger]
root=DEBUG

iphizic, 2024-11-20:
Where nfc-spy.log located. I can't find him, and when I close application, see segmentation fault in term.

josevcm, 2024-11-20:
In user home, under ".local/share/josevcm/nfc-spy/log/nfc-spy.log"

iphizic, 2024-11-20:
[nfc-spy.log](https://github.com/user-attachments/files/17828062/nfc-spy.log)

josevcm, 2024-11-20:
Please change gainMode=1 an set 12db aprox

josevcm, 2024-11-23:
I close the issue, if continue the problem, please open a new one.

closed 2024-11-23 by josevcm