josevcm/nfc-laboratory #46: Compiler warnings deprecated-declarations

Steffen-W, 2024-11-19:
The code can be compiled and runs without any problems. However, the compiler issues the following warnings. However, these should be easy to fix. If you like, I can also make a merge request.

```make
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp: In member function 'void HexViewWidget::Impl::layout()':
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp:121:36: warning: 'qsizetype QByteArray::count() const' is deprecated: Use size() or length() instead. [-Wdeprecated-declarations]
  121 |       int dataHeight = ((data.count() / lineBytes) + (data.count() % lineBytes ? 1 : 0)) * charHeight;
      |                          ~~~~~~~~~~^~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp:121:65: warning: 'qsizetype QByteArray::count() const' is deprecated: Use size() or length() instead. [-Wdeprecated-declarations]
  121 |       int dataHeight = ((data.count() / lineBytes) + (data.count() % lineBytes ? 1 : 0)) * charHeight;
      |                                                       ~~~~~~~~~~^~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp:129:32: warning: 'qsizetype QByteArray::count() const' is deprecated: Use size() or length() instead. [-Wdeprecated-declarations]
  129 |       if (lastLine > data.count() / lineBytes)
      |                      ~~~~~~~~~~^~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp:130:32: warning: 'qsizetype QByteArray::count() const' is deprecated: Use size() or length() instead. [-Wdeprecated-declarations]
  130 |          lastLine = (data.count() / lineBytes) + (data.count() % lineBytes ? 1 : 0);
      |                      ~~~~~~~~~~^~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp:130:61: warning: 'qsizetype QByteArray::count() const' is deprecated: Use size() or length() instead. [-Wdeprecated-declarations]
  130 |          lastLine = (data.count() / lineBytes) + (data.count() % lineBytes ? 1 : 0);
      |                                                   ~~~~~~~~~~^~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp: In member function 'void HexViewWidget::Impl::paint(QPaintEvent*)':
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp:149:96: warning: 'qsizetype QByteArray::count() const' is deprecated: Use size() or length() instead. [-Wdeprecated-declarations]
  149 |       for (int addr = firstLine * lineBytes, line = firstLine, lineCoord = 0; addr < data.count() && line <= lastLine; addr += lineBytes, lineCoord += charHeight, line++)
      |                                                                                      ~~~~~~~~~~^~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp:159:86: warning: 'qsizetype QByteArray::count() const' is deprecated: Use size() or length() instead. [-Wdeprecated-declarations]
  159 |          for (int i = 0, pos = addr, charCoord = 0; i < lineBytes && pos < data.count(); i++, pos++, charCoord += charWidth * 3)
      |                                                                            ~~~~~~~~~~^~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp:173:86: warning: 'qsizetype QByteArray::count() const' is deprecated: Use size() or length() instead. [-Wdeprecated-declarations]
  173 |          for (int i = 0, pos = addr, charCoord = 0; i < lineBytes && pos < data.count(); i++, pos++, charCoord += charWidth)
      |                                                                            ~~~~~~~~~~^~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp: In static member function 'static QString HexViewWidget::Impl::toHexString(const QByteArray&, int, int)':
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp:204:41: warning: 'qsizetype QByteArray::count() const' is deprecated: Use size() or length() instead. [-Wdeprecated-declarations]
  204 |       for (int i = from; i < value.count() && i < to; i++)
      |                              ~~~~~~~~~~~^~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp: In static member function 'static QString HexViewWidget::Impl::toAsciiString(const QByteArray&, int, int)':
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\widgets\HexViewWidget.cpp:216:41: warning: 'qsizetype QByteArray::count() const' is deprecated: Use size() or length() instead. [-Wdeprecated-declarations]
  216 |       for (int i = from; i < value.count() && i < to; i++)
      |                              ~~~~~~~~~~~^~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\3party\customplot\QCustomPlot.cpp: In member function 'virtual QString QCPAxisTickerDateTime::getTickLabel(double, const QLocale&, QChar, int)':
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\3party\customplot\QCustomPlot.cpp:6726:58: warning: 'QDateTime QDateTime::toTimeSpec(Qt::TimeSpec) const' is deprecated: Use toTimeZone instead [-Wdeprecated-declarations]
 6726 |     return locale.toString(keyToDateTime(tick).toTimeSpec(mDateTimeSpec), mDateTimeFormat);
      |                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\3party\customplot\QCustomPlot.cpp: In static member function 'static double QCPAxisTickerDateTime::dateTimeToKey(const QDate&, Qt::TimeSpec)':
nfc-laboratory\src\nfc-app\app-qt\src\main\cpp\3party\customplot\QCustomPlot.cpp:6831:25: warning: 'QDateTime QDate::startOfDay(Qt::TimeSpec, int) const' is deprecated: Pass QTimeZone instead [-Wdeprecated-declarations]
 6831 |   return date.startOfDay(timeSpec).toMSecsSinceEpoch()/1000.0;
      |          ~~~~~~~~~~~~~~~^~~~~~~~~~
```

I would be really happy if you could create a release of the current version. The code you have created is awesome!

josevcm, 2024-11-19:
Yes, please make a pull request for this, my compiler does not trigger such warnings.

I am trying to configure GitHub actions to generate Windows and Linux releases but it is unknown to me and as you have seen in my last commits I use trial and error...

Thank you very much for your support!

Steffen-W, 2024-11-19:
No problem. I am also trying to solve the problem. Maybe it will work now. :)

https://github.com/Steffen-W/nfc-laboratory/actions/runs/11920049074/job/33221122119

josevcm, 2024-12-02:
Closed as do not reproduce under CI/CD build with github actions.

closed 2024-12-02 by josevcm