josevcm/nfc-laboratory #60: nfc-rx dont work

Steffen-W, 2025-03-17:
```bash
$ cmake -DCMAKE_BUILD_TYPE=Debug -S nfc-laboratory -B build
...
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found WrapAtomic: TRUE  
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found WrapOpenGL: TRUE  
-- Found XKB: /usr/lib/x86_64-linux-gnu/libxkbcommon.so (found suitable version "1.6.0", minimum required is "0.5.0") 
-- Found WrapVulkanHeaders: /usr/include  
-- Configuring done (0.9s)
-- Generating done (0.1s)
-- Build files have been written to: .../build

$ cmake --build build --target all -- -j$(nproc)
[  3%] Built target mufft-sse
[  3%] Built target microtar
[  3%] Built target mufft-sse3
[  7%] Built target mirisdr
[  7%] Built target mufft-avx
[ 14%] Built target ed25519
[ 16%] Built target mufft
[ 22%] Built target rt-lang
[ 25%] Built target hw-dev
[ 26%] Built target hw-logic
[ 29%] Built target hw-radio
[ 31%] Built target lab-data
[ 32%] Built target test-dio
[ 35%] Built target lab-logic
[ 39%] Built target lab-radio
[ 41%] Built target test-sdr
[ 47%] Built target lab-tasks
[ 47%] Built target nfc-lab_autogen_timestamp_deps
[ 48%] Built target nfc-rx
[ 49%] Built target nfc-lab_autogen
[100%] Built target nfc-lab

$ build/src/nfc-app/app-rx/nfc-rx -v
2025-03-17 12:54:51.691 INFO [136772857870208] (app.main) NFC laboratory, 2024 Jose Vicente Campos Martinez
2025-03-17 12:54:51.691 INFO [136772841375424] (rt.Executor) task FrameDecoderTask started in thread -392165696
2025-03-17 12:54:51.691 INFO [136772841375424] (rt.Worker) started worker FrameDecoderTask
2025-03-17 12:54:51.691 INFO [136772820403904] (rt.Executor) task RadioDeviceTask started in thread -413137216
2025-03-17 12:54:51.691 INFO [136772820403904] (rt.Worker) started worker RadioDeviceTask
2025-03-17 12:54:51.691 INFO [136772820403904] (worker.RadioDevice) registering devices
2025-03-17 12:54:51.691 INFO [136772820403904] (hw.DeviceFactory) registering device type radio.airspy
2025-03-17 12:54:51.691 INFO [136772820403904] (hw.DeviceFactory) registering device type radio.rtlsdr
2025-03-17 12:54:51.691 INFO [136772820403904] (hw.DeviceFactory) registering device type radio.miri
2025-03-17 12:54:51.691 INFO [136772841375424] (worker.RadioDecoder) status update: {"debugEnabled":false,"powerLevelThreshold":0.009999999776482582,"queueSize":0,"sampleRate":0,"sampleThroughput":0.0,"status":"disabled","streamTime":0}
```

The gui works perfectly:
```
./nfc-lab
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Disabled direct sampling mode
[R82XX] PLL not locked!
```

What do I have to change to get the nfc data packets displayed in the terminal?

Steffen-W, 2025-03-22:
I had created a parser for the live interpretation and made an output for this in the terminal. Not perfect but it works. Where would be a suitable place to output the packets? The GUI parallel makes it easier to align the coil correctly. I would still be in favor of being able to specify something like -v for the parallel playback of the packets when starting via the terminal.

[StreamModel.cpp#L214-L243](https://github.com/josevcm/nfc-laboratory/blob/master/src/nfc-app/app-qt/src/main/cpp/model/StreamModel.cpp#L214-L243)

```cpp
   QVariant dataValue(const QModelIndex &index, const lab::RawFrame &frame, const lab::RawFrame &prev) const
   {
      // Statische Variablen für jeden Wert, die über Funktionsaufrufe hinweg bestehen bleiben
      static QVariant idValue;
      static QVariant timeValue;
      static QVariant deltaValue;
      static QVariant rateValue;
      static QVariant techValue;
      static QVariant eventValue;
      static QVariant flagsValue;
      static QVariant dataValueVar; // Für den Data-Wert
      static QVariant maxTimeValue; // Variable für die aktuellste Zeit

      QVariant value; // Lokale Variable für den aktuellen Rückgabewert
      QByteArray hexData;

      switch (index.column())
      {
      case Id:
         idValue = index.row();
         value = idValue;
         break;

      case Time:
         timeValue = timeSource == Elapsed ? frameTime(frame) : frameStart(frame);
         value = timeValue;
         break;

      case Delta:
         deltaValue = frameDelta(frame, prev);
         value = deltaValue;
         break;

      case Rate:
         rateValue = frameRate(frame);
         value = rateValue;
         break;

      case Tech:
         techValue = frameTech(frame);
         value = techValue;
         break;

      case Event:
         eventValue = frameEvent(frame, prev);
         value = eventValue;
         break;

      case Flags:
         flagsValue = frameFlags(frame);
         value = flagsValue;
         break;

      case Data:
         dataValueVar = frameData(frame);
         value = dataValueVar;
         hexData = dataValueVar.toByteArray();
         if (!hexData.isEmpty() && (maxTimeValue.isNull() || timeValue.toDouble() > maxTimeValue.toDouble()))
         {
            maxTimeValue = timeValue; // Aktuellste Zeit aktualisieren
            printf("{Id: '%s', ", idValue.toString().toStdString().c_str());
            printf("Time: '%s', ", timeValue.toString().toStdString().c_str());
            printf("Delta: '%s', ", deltaValue.toString().toStdString().c_str());
            printf("Rate: '%s', ", rateValue.toString().toStdString().c_str());
            printf("Tech: '%s', ", techValue.toString().toStdString().c_str());
            printf("Event: '%s', ", eventValue.toString().toStdString().c_str());
            printf("Flags: '%s', ", flagsValue.toString().toStdString().c_str());
            // printf("Data: '%s'}\n", hexData.toStdString().c_str());
            printf("Data: '%s'}\n", hexData.toHex().constData());
         }
         break;

      default:
         break;
      }

      return value;
   }
```

josevcm, 2025-03-22:
Sorry Steffen, I haven't had time to look at it. What do you mean by the GUI in parallel?

Steffen-W, 2025-03-22:
No problem. Yes, I mean parallel to the GUI, because without the GUI it is not too easy to notice a misalignment of the coil.

josevcm, 2025-04-11:
Hi Steffen

Please build new version for nfc-rx.exe and check if is valid for you:

> nfc-rx.exe -f 27120000 -s 3200000
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
Disabled direct sampling mode
receiver configuration: {"gainMode":1,"gainValue":77}
decoder configuration: {"sampleRate":3200000}
000000.000 (CarrierOff)
000000.000 (CarrierOn)
000000.765 (PCD->PICC) [NfcA@106]: 26
000000.805 (PCD->PICC) [NfcA@106]: 26
000000.810 (PCD->PICC) [NfcA@106]: 26
000001.114 (PCD->PICC) [NfcA@106]: 26
000001.120 (PCD->PICC) [NfcA@106]: 26
000001.159 (PCD->PICC) [NfcA@106]: 26
000001.164 (PCD->PICC) [NfcA@106]: 26
000001.468 (PCD->PICC) [NfcA@106]: 26
000001.474 (PCD->PICC) [NfcA@106]: 26
000001.518 (PCD->PICC) [NfcA@106]: 26
000001.822 (PCD->PICC) [NfcA@106]: 26
000001.867 (PCD->PICC) [NfcA@106]: 26
000001.873 (PCD->PICC) [NfcA@106]: 26
000002.183 (PCD->PICC) [NfcA@106]: 26
000002.227 (PCD->PICC) [NfcA@106]: 26
000002.531 (PCD->PICC) [NfcA@106]: 26
000002.537 (PCD->PICC) [NfcA@106]: 26
000002.576 (PCD->PICC) [NfcA@106]: 26
000002.582 (PCD->PICC) [NfcA@106]: 26
000002.886 (PCD->PICC) [NfcA@106]: 26
000002.891 (PCD->PICC) [NfcA@106]: 26
000002.930 (PCD->PICC) [NfcA@106]: 26
000002.936 (PCD->PICC) [NfcA@106]: 26
000003.240 (PCD->PICC) [NfcA@106]: 26
000003.246 (PCD->PICC) [NfcA@106]: 26
000003.285 (PCD->PICC) [NfcA@106]: 26
000003.291 (PCD->PICC) [NfcA@106]: 26
000003.595 (PCD->PICC) [NfcA@106]: 26
000003.600 (PCD->PICC) [NfcA@106]: 26
000003.640 (PCD->PICC) [NfcA@106]: 26
000003.949 (PCD->PICC) [NfcA@106]: 26
000003.950 (PCD->PICC) [NfcA@106]: 33
000003.952 (PCD->PICC) [NfcA@106]: 93 70 DE D6 AD 36 93 39 40
000004.216 (PCD->PICC) [NfcA@106]: 52
000004.222 (PCD->PICC) [NfcA@106]: 52
000004.223 (PCD->PICC) [NfcA@106]: 93 70 DE D6 AD 36 73 27
000004.369 (PCD->PICC) [NfcA@106]: 52
000004.374 (PCD->PICC) [NfcA@106]: 52
000004.376 (PCD->PICC) [NfcA@106]: 93 70 DE D6 AD 36 93 39 40
000004.511 (PCD->PICC) [NfcA@106]: 52
000004.518 (PCD->PICC) [NfcA@106]: 93 70 DE 56 95 9C CE 04
000004.782 (PCD->PICC) [NfcA@106]: 52
000004.789 (PCD->PICC) [NfcA@106]: 93 70 BE 5A 95 E6 72 27
000005.052 (PCD->PICC) [NfcA@106]: 52
000005.058 (PCD->PICC) [NfcA@106]: 52
000005.059 (PCD->PICC) [NfcA@106]: 12 C3
000005.060 (PCD->PICC) [NfcA@106]: 42 21 F2
000005.323 (PCD->PICC) [NfcA@106]: 52
000005.329 (PCD->PICC) [NfcA@106]: 52
000005.330 (PCD->PICC) [NfcA@106]: 93 70 DE D6 AD 36 93 39 40
000005.594 (PCD->PICC) [NfcA@106]: 52
000005.600 (PCD->PICC) [NfcA@106]: 52
000005.601 (PCD->PICC) [NfcA@106]: 93 B0 BB 5A 95 E6 72 27
000005.870 (PCD->PICC) [NfcA@106]: 52
000005.887 (PCD->PICC) [NfcA@106]: 52
000005.892 (PCD->PICC) [NfcA@106]: 52
000006.161 (PCD->PICC) [NfcA@106]: 26
000006.200 (PCD->PICC) [NfcA@106]: 26
000006.510 (PCD->PICC) [NfcA@106]: 26
000006.516 (PCD->PICC) [NfcA@106]: 26
000006.555 (PCD->PICC) [NfcA@106]: 26
000006.560 (PCD->PICC) [NfcA@106]: 26
000006.865 (PCD->PICC) [NfcA@106]: 26
000006.870 (PCD->PICC) [NfcA@106]: 26
000006.909 (PCD->PICC) [NfcA@106]: 26
000006.915 (PCD->PICC) [NfcA@106]: 26
000007.219 (PCD->PICC) [NfcA@106]: 26
000007.225 (PCD->PICC) [NfcA@106]: 26
000007.264 (PCD->PICC) [NfcA@106]: 26
000007.270 (PCD->PICC) [NfcA@106]: 26

closed 2025-04-11 by josevcm

Steffen-W, 2025-04-16:
Thank you very much for your work. Unfortunately, I do not get any data packages displayed. :( but nfc-lab works perfectly for me.

```
./build/src/nfc-app/app-rx/nfc-rx -v
2025-04-16 13:22:08.258 INFO [123906068055936] (app.main) NFC laboratory, 2024 Jose Vicente Campos Martinez
2025-04-16 13:22:08.259 INFO [123906037909184] (rt.Executor) task FrameDecoderTask started in thread 526386880
2025-04-16 13:22:08.259 INFO [123906037909184] (rt.Worker) started worker FrameDecoderTask
2025-04-16 13:22:08.259 INFO [123906016937664] (rt.Executor) task RadioDeviceTask started in thread 505415360
2025-04-16 13:22:08.259 INFO [123906016937664] (rt.Worker) started worker RadioDeviceTask
2025-04-16 13:22:08.259 INFO [123906016937664] (worker.RadioDevice) registering devices
2025-04-16 13:22:08.259 INFO [123906037909184] (worker.RadioDecoder) status update: {"debugEnabled":false,"powerLevelThreshold":0.009999999776482582,"queueSize":0,"sampleRate":0,"sampleThroughput":0.0,"status":"disabled","streamTime":0}
2025-04-16 13:22:08.259 INFO [123906016937664] (hw.DeviceFactory) registering device type radio.airspy
2025-04-16 13:22:08.259 INFO [123906016937664] (hw.DeviceFactory) registering device type radio.rtlsdr
2025-04-16 13:22:08.259 INFO [123906016937664] (hw.DeviceFactory) registering device type radio.miri

```

Steffen-W, 2025-04-16:
Sorry was probably my mistake I was still on the main branch. 
`git checkout develop`
`cmake -DCMAKE_BUILD_TYPE=Debug -S nfc-laboratory -B build`
`cmake --build build --target all -- -j$(nproc)`

```
./build/src/nfc-app/app-rx/nfc-rx 
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Disabled direct sampling mode
[R82XX] PLL not locked!
receiver configuration: {"gainMode":1,"gainValue":77}
decoder configuration: {"sampleRate":3200000}
*** buffer overflow detected ***: terminated
```

Unfortunately, the code crashes very quickly due to a memory Overflow.

```
./build/src/nfc-app/app-rx/nfc-rx -v
2025-04-16 13:34:30.119 INFO [130871850444672] (app.main) NFC laboratory, 2024 Jose Vicente Campos Martinez
2025-04-16 13:34:30.119 INFO [130871820551872] (rt.Executor) task FrameDecoderTask started in thread -127924544
2025-04-16 13:34:30.119 INFO [130871820551872] (rt.Worker) started worker FrameDecoderTask
2025-04-16 13:34:30.119 INFO [130871820551872] (worker.RadioDecoder) status update: {"debugEnabled":false,"powerLevelThreshold":0.009999999776482582,"queueSize":0,"sampleRate":0,"sampleThroughput":0.0,"status":"disabled","streamTime":0}
2025-04-16 13:34:30.119 INFO [130871820551872] (worker.RadioDecoder) change config: {"enabled":true}
2025-04-16 13:34:30.119 INFO [130871810066112] (rt.Executor) task RadioDeviceTask started in thread -138410304
2025-04-16 13:34:30.119 INFO [130871810066112] (rt.Worker) started worker RadioDeviceTask
2025-04-16 13:34:30.119 INFO [130871810066112] (worker.RadioDevice) registering devices
2025-04-16 13:34:30.119 INFO [130871810066112] (hw.DeviceFactory) registering device type radio.airspy
2025-04-16 13:34:30.119 INFO [130871810066112] (hw.DeviceFactory) registering device type radio.rtlsdr
2025-04-16 13:34:30.119 INFO [130871810066112] (hw.DeviceFactory) registering device type radio.miri
2025-04-16 13:34:30.119 INFO [130871810066112] (worker.RadioDevice) change config: {"enabled":true}
2025-04-16 13:34:30.119 INFO [130871810066112] (worker.RadioDevice) status update: {"status":"absent"}
2025-04-16 13:34:30.119 INFO [130871820551872] (worker.RadioDecoder) status update: {"debugEnabled":false,"powerLevelThreshold":0.009999999776482582,"protocol":{"nfca":{"correlationThreshold":0.75,"enabled":true,"maximumModulationDeep":1.0,"minimumModulationDeep":0.8999999761581421},"nfcb":{"correlationThreshold":0.5,"enabled":true,"maximumModulationDeep":0.8999999761581421,"minimumModulationDeep":0.10000000149011612},"nfcf":{"correlationThreshold":0.5,"enabled":true,"maximumModulationDeep":0.8999999761581421,"minimumModulationDeep":0.10000000149011612},"nfcv":{"correlationThreshold":0.5,"enabled":true,"maximumModulationDeep":1.0,"minimumModulationDeep":0.8999999761581421}},"queueSize":0,"sampleRate":0,"sampleThroughput":0.0,"status":"idle","streamTime":0}
2025-04-16 13:34:30.159 INFO [130871810066112] (worker.RadioDevice) detected device radio.rtlsdr://00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Disabled direct sampling mode
[R82XX] PLL not locked!
2025-04-16 13:34:31.141 INFO [130871810066112] (hw.RealtekDevice) opened rtlsdr device radio.rtlsdr://00000001 width tuner type 5
2025-04-16 13:34:31.141 INFO [130871810066112] (worker.RadioDevice) device radio.rtlsdr://00000001 connected!
2025-04-16 13:34:31.141 INFO [130871810066112] (worker.RadioDevice) applying configuration: {"enabled":true}
2025-04-16 13:34:31.141 INFO [130871810066112] (worker.RadioDevice) status update: {"biasTee":0,"centerFreq":27120000,"directSampling":0,"gainMode":0,"gainModes":[{"name":"Auto","value":0},{"name":"Manual","value":1}],"gainValue":0,"gainValues":[{"name":"0.00 db","value":0},{"name":"0.90 db","value":9},{"name":"1.40 db","value":14},{"name":"2.70 db","value":27},{"name":"3.70 db","value":37},{"name":"7.70 db","value":77},{"name":"8.70 db","value":87},{"name":"12.50 db","value":125},{"name":"14.40 db","value":144},{"name":"15.70 db","value":157},{"name":"16.60 db","value":166},{"name":"19.70 db","value":197},{"name":"20.70 db","value":207},{"name":"22.90 db","value":229},{"name":"25.40 db","value":254},{"name":"28.00 db","value":280},{"name":"29.70 db","value":297},{"name":"32.80 db","value":328},{"name":"33.80 db","value":338},{"name":"36.40 db","value":364},{"name":"37.20 db","value":372},{"name":"38.60 db","value":386},{"name":"40.20 db","value":402},{"name":"42.10 db","value":421},{"name":"43.40 db","value":434},{"name":"43.90 db","value":439},{"name":"44.50 db","value":445},{"name":"48.00 db","value":480},{"name":"49.60 db","value":496}],"mixerAgc":0,"model":"RTLSDR","name":"radio.rtlsdr://00000001","sampleRate":3200000,"sampleRates":[{"name":"225000","value":225000},{"name":"900000","value":900000},{"name":"1024000","value":1024000},{"name":"1400000","value":1400000},{"name":"1800000","value":1800000},{"name":"1920000","value":1920000},{"name":"2048000","value":2048000},{"name":"2400000","value":2400000},{"name":"2560000","value":2560000},{"name":"2800000","value":2800000},{"name":"3200000","value":3200000}],"samplesLost":0,"samplesRead":0,"serial":"00000001","status":"idle","streamTime":0,"tunerAgc":0,"vendor":"Generic","version":""}
receiver configuration: {"gainMode":1,"gainValue":77}
decoder configuration: {"sampleRate":3200000}
2025-04-16 13:34:31.620 INFO [130871820551872] (worker.RadioDecoder) change config: {"sampleRate":3200000}
[... cut ...]
2025-04-16 13:34:32.149 INFO [130871810066112] (worker.RadioDevice) status update: {"biasTee":0,"centerFreq":27120000,"directSampling":0,"gainMode":1,"gainValue":77,"mixerAgc":0,"model":"RTLSDR","name":"radio.rtlsdr://00000001","sampleRate":3200000,"samplesLost":0,"samplesRead":0,"serial":"00000001","status":"streaming","streamTime":1744803272,"tunerAgc":0,"vendor":"Generic","version":""}
2025-04-16 13:34:32.149 INFO [130871789094592] (hw.RealtekDevice) stream worker started for device radio.rtlsdr://00000001
*** buffer overflow detected ***: terminated
```

josevcm, 2025-04-17:
It seems to be a problem in Linux, I tried it with Windows, I will review it as soon as I can, sorry