josevcm/nfc-laboratory #61: Can't change receiveing frequency from harmonic to actual 13

ccloned, 2025-04-05:
I've altered nfc-lab.ini as follows:
[device.radio.rtlsdr]
centerFreq=13560000
sampleRate=2400000
gainMode=1
gainValue=125
biasTee=1
directSampling=0
mixerAgc=0
tunerAgc=0

Particularly:
centerFreq=13560000
sampleRate=2400000

...to use my RTL-SDR V4 dongle which can operate at native NFC freq.
However, software doesn't react to that change and still listens to 27.12 MHz, which is suboptimal to my use-case.
Is that intentional?

josevcm, 2025-04-05:
Hi, in witch folder are you change the nfc-lab.ini? and you are using latest release or build yourself from master branch?

ccloned, 2025-04-05:
I've used the latest release, and I change it in config folder

ccloned, 2025-04-05:
Oh no. I have to get some sleep. I'm sorry. Should've changed those in %appdata%
I'm sorry and thank you for responding.

closed 2025-04-05 by ccloned

josevcm, 2025-04-05:
I hope it is useful to you, if you need anything else please contact me